kaosat-dev/CoffeeSCad #157: Question

berserk82, 2017-05-12:
Does COffeeSCad require anything but a webserver to run? (or does it need node, frameworks, etc)

kaosat-dev, 2017-05-12:
Hello @berserk82 , sorry but the project is pretty much dead. I think you could just build the 'distributable' file once using node and use any webserver to run, there is nothing fancy in there, but the whole tooling and everything else is very outdated, so not sure any of it still works.
Please try https://openjscad.org/ instead, it is quite similar, and well maintained (I am also working on it)

berserk82, 2017-05-12:
I'm interested in no-node-required apps

kaosat-dev, 2017-05-12:
@berserk82 for OpenJSCAD (and back then for CoffeeSCAD), Node.js is only needed for some very specific development OF the software itself, you can host your designs running in OpenJSCAD without Node.js or anything else, it is just a simple html/js file

berserk82, 2017-05-12:
Ah, ok, thanks. I'll look at OpenJsCad then.

closed 2017-05-12 by berserk82